jjm2473-org/openwrta #1: 请问如何同时开启SMB和NFS共享

boyu899, 2020-08-27:
请问如何同时开启SMB和NFS共享 因为家里现在老的X9S刷了最新openwrt 新的RT1619主控Z9X 只能选择SMB和NFS(4K原盘SMB速度缓冲卡顿) 有没有教程同时开启SMB(日常共享使用)和NFS(4K原盘播放使用)

boyu899, 2020-08-27:
另外同时设置FTP和SMB是可以的.但是重启X9S后 其它设备再次访问就空白了 可以连接 文件看不到  Z9X要第三方软件使用FTP 比较麻烦 请大神指点X9S同时开启SMB和NFS共享一个硬盘的目录

jjm2473, 2020-08-28:
https://github.com/jjm2473/rtd1295-openwrt18-porting/blob/z9s_luci/docs/porting.md#%E7%BD%91%E7%BB%9C%E5%85%B1%E4%BA%AB%E7%9A%84%E9%99%90%E5%88%B6

jjm2473, 2020-08-28:
FTP共享/mnt应该没问题的，共享挂载点的条件是挂载点对匿名FTP用户不可写，不然VSFTPD拒绝共享。
NFS只能共享挂载点（例如/mnt/sda1），应该是内核安全限制。

jjm2473, 2020-08-28:
Linux 4.16才开始支持NFS导出Overlayfs。
但是我们的内核是4.1，所以不支持。

https://cateee.net/lkddb/web-lkddb/OVERLAY_FS_NFS_EXPORT.html

jjm2473, 2020-08-28:
这里整理了更加详细的情况分析 

https://github.com/jjm2473/openwrt/wiki/RTD129X-%E7%BD%91%E7%BB%9C%E5%85%B1%E4%BA%AB%E7%9B%B8%E5%85%B3%E9%97%AE%E9%A2%98-%EF%BC%88Samba%EF%BC%8CNFS%EF%BC%8CFTP%EF%BC%89

jjm2473, 2020-08-28:
重启以后不能访问的问题，可能是因为根本没挂载成功，检查下磁盘是不是挂载了。如果硬盘太大，通常会出现来不及挂载的情况。

jjm2473, 2020-08-28:
4K原盘的码率是多少，我之前测试SMB通常比NFS快，但是现在网口只有100Mbps，没办法测试

boyu899, 2020-08-28:
> 4K原盘的码率是多少，我之前测试SMB通常比NFS快，但是现在网口只有100Mbps，没办法测试

4k原盘码率80m 双子杀手95m  电脑-x9s smb传输没问题 稳稳的112m 满千兆  安卓设备smb应该是有瓶顶 40m左右 播放会偶尔卡顿 yongftp完全流畅 就是要要第三方 es文件app才能使用ftp连接x9s  请大神出个教程 同时nfs和smb 针对zidoo互通

boyu899, 2020-08-28:
nfs和smb同时共享挂载点/mnt/sda1  能连接 不重启前smb依旧可以 nfs连接空白 重启后两个协议都空白文件夹

jjm2473, 2020-08-28:
刚刚测试了一下，Mac通过5Gwifi链接路由器，路由器有线连接Z9S，测试Mac到Z9S的网速是50-60M/s，测试Samba速度是40M/s，测试NFS速度是36M/s。

jjm2473, 2020-08-28:
重启以后空白的问题，前面已经说过了，检查磁盘开机时是不是挂载成功了。我的硬盘只有1T，所以不会出现挂载失败的问题。

jjm2473, 2020-08-28:
nfs连接空白的问题，看看系统日志，一般会告诉你为什么共享失败

jjm2473, 2020-08-28:
系统日志在 http://设备IP/cgi-bin/luci/admin/status/syslog

boyu899, 2020-08-28:
好的 谢谢 我再捣鼓捣鼓 也请大神有空做个教程 随手笔记下 更方便我们这些小白

jjm2473, 2020-08-28:
嗯，刚刚已经把注意事项写在wiki里了

boyu899, 2020-08-30:
最新版本的固件设置好了 ftp三个用户不能共享同一个文件夹 我的设置方法 smb共享mnt ftp本地用户共享mnt 虚拟用户共享mnt/sda1  匿名用户还有点不懂承接用户权限 尝试过两次 重启都空白无权限 最后只开启了本地和虚拟添加了一个自定义用户和密码 可以了 重启后也不会空白

jjm2473, 2020-08-30:
重启空白跟用户设置应该没什么关系。匿名用户选择ftp，可以共享/mnt

boyu899, 2020-08-31:
又到捣鼓NFS的时候 没办法 芝杜盒子要用FTP只能第三方软件 目前SMB和FTP基本完美 没有任何问题  添加NFS 无论设置你在wiki下说明的任何文件 打开都是空白 设置的是  /mnt/sda1或里面的movie文件夹(FTP有共享这个文件夹) 允许用户指定了Z9X的IP 权限默认固件上的权限  一直空白 显示不出  再有个问题芝杜NFS扫描默认111端口 是否要修改哪个地方之类 附上SSH操作

root@OpenWrt:~# mount t nfs 10.0.0.10:/sda1 /mnt
mount: bad usage
Try 'mount --help' for more information.
root@OpenWrt:~# /etc/init.d/portmap start
-ash: /etc/init.d/portmap: not found
root@OpenWrt:~# /etc/init.d/nfs start
root@OpenWrt:~# /etc/init.d/portmap start
-ash: /etc/init.d/portmap: not found
root@OpenWrt:~# /etc/init.d/nfs start
root@OpenWrt:~# mount t nfs 10.0.0.10:/sda1 /mnt
mount: bad usage
Try 'mount --help' for more information.
root@OpenWrt:~# #vi /etc/exports
root@OpenWrt:~# #vi /etc/exports
root@OpenWrt:~# vi /etc/exports
root@OpenWrt:~# iptables
iptables v1.6.2: no command specified
Try `iptables -h' or 'iptables --help' for more information.
root@OpenWrt:~# ipchains
-ash: ipchains: not found
root@OpenWrt:~# /etc/hosts.deny
-ash: /etc/hosts.deny: not found
root@OpenWrt:~# /etc/hosts.deny
-ash: /etc/hosts.deny: not found
root@OpenWrt:~# iptables
iptables v1.6.2: no command specified
Try `iptables -h' or 'iptables --help' for more information.
root@OpenWrt:~# #vi /etc/exports
root@OpenWrt:~# vi /etc/exports
root@OpenWrt:~# root@OpenWrt:~#

boyu899, 2020-08-31:
注明下 设置用的openwrt固件里面的nfs插件设置 不是ssh ssh是乱捣鼓的 可以不研究

只修改成/mnt/sda1  /mnt/sda1/movie 奇怪的是设置了/mnt/sda1/*   z9x扫描连接上nfs进去sda1打开还真能显示出个*文件夹 打开空白

jjm2473, 2020-09-02:
你的问题不是很清楚，看你的操作也目的不明，你是打算在我的固件共享NFS还是挂载NFS？
如果你是共享，根本用不上`mount t nfs`，就算是挂载，也应该是`mount -t nfs`（`t`前面少了一个`-`）
接下来我只讨论我的固件中的情况，怎么在其他设备挂载NFS另外讨论。
portmap早已经被弃用了，现在使用rpcbind，而且一般情况下也不用自己启动rpcbind。
我在固件里面默认禁用了防火墙，所以iptables一般也不需要执行。

执行这些命令，把输出都发给我看看：
```
ps
mount
cat /etc/config/nfs
exportfs -a
logread | tail -10
exportfs -s
```

jjm2473, 2020-09-02:
如果你要挂载NFS，参考 https://github.com/jjm2473/rtd1295-openwrt18-porting/wiki/%E6%8C%82%E8%BD%BDSMB(CIFS),-NFS

boyu899, 2020-09-02:
root@OpenWrt:~# ps
  PID USER       VSZ STAT COMMAND
    1 root      1592 S    /sbin/procd
   95 root      1140 S    /sbin/ubusd
  114 root       816 S    /sbin/xubus
  226 root      1188 S    /sbin/logd -S 64
  245 root      1716 S    /sbin/rpcd -s /var/run/ubus.sock -t 30
  265 root      1232 S    /sbin/mdadm --monitor --syslog --scan --config=/var/etc/mdadm.conf
  300 root       968 S    /usr/sbin/dropbear -F -P /var/run/dropbear.1.pid -p 22 -K 300 -T 3
  323 rpc       1260 S    /usr/sbin/rpcbind -f -w
  356 root      1680 S    /sbin/netifd
  382 root      1448 S    /usr/sbin/odhcpd
  508 root      2872 S    /usr/sbin/vsftpd -olisten_address=0.0.0.0 -olisten=YES -olisten_ipv6=NO /var/run/vsftpd/vsft
  540 root      1368 S    sh /lib/netifd/dns.script reload -i eth0
  554 root     20336 S    /usr/bin/minidlna -f /tmp/minidlna.conf
  581 root      1972 S    /usr/sbin/uhttpd -f -h /www -r OpenWrt -x /cgi-bin -t 60 -T 30 -k 20 -A 1 -n 3 -N 100 -R -p
  593 root      1536 S    /usr/sbin/dbus-daemon --system
  610 nobody    2384 S    avahi-daemon: running [OpenWrt.local]
  874 root      1196 S<   /usr/sbin/ntpd -n -N -S /usr/sbin/ntpd-hotplug -p 0.openwrt.pool.ntp.org -p 1.openwrt.pool.n
  927 root      2204 S    /usr/sbin/netatalk -d -F /etc/afp.conf
  957 nfs       1564 S    /usr/sbin/rpc.statd -p 32778 -o 32779 -F
  958 root      2056 S    /usr/sbin/rpc.mountd -p 32780 -F
  996 root      2632 S    /usr/sbin/afpd -d -F /etc/afp.conf
  997 root      1612 S    /usr/sbin/cnid_metad -d -F /etc/afp.conf
 1131 root      1196 S    lock /var/lock/dns_eth0.lock
 1132 root      1196 S    lock -w /var/lock/dns_eth0.lock
 1209 root       852 S    /usr/bin/wsdd2 -i eth0 -N OpenWrt -G WORKGROUP -b vendor:Realtek,model:Realtek_RTD1295,sku:F
 1292 root     33612 S    /usr/sbin/smbd -F
 1293 root     20112 S    /usr/sbin/nmbd -F
 1305 root     32320 S    {smbd-notifyd} /usr/sbin/smbd -F
 1306 root     32328 S    {cleanupd} /usr/sbin/smbd -F
 7428 root     34032 S    /usr/sbin/smbd -F
 7437 root      1092 R    /usr/sbin/dropbear -F -P /var/run/dropbear.1.pid -p 22 -K 300 -T 3
 7438 root      1204 S    -ash
 7454 root      1200 R    ps
root@OpenWrt:~# mount
/dev/mmcblk0p9 on /rom type squashfs (ro,relatime)
tmpfs on /rom/android type tmpfs (ro,noatime,seclabel,size=10240k)
proc on /proc type proc (rw,nosuid,nodev,noexec,noatime)
sysfs on /sys type sysfs (rw,nosuid,nodev,noexec,noatime,seclabel)
cgroup on /sys/fs/cgroup type cgroup (rw,nosuid,nodev,noexec,relatime,cpuset,cpu,cpuacct,blkio,memory,devices,freezer,net_cls,net_prio)
tmpfs on /tmp type tmpfs (rw,nosuid,nodev,noatime,seclabel)
/dev/loop0 on /overlay type ext4 (rw,relatime,seclabel,data=ordered)
overlayfs:/overlay on / type overlay (rw,noatime,lowerdir=/,upperdir=/overlay/upper,workdir=/overlay/work)
tmpfs on /dev type tmpfs (rw,nosuid,relatime,seclabel,size=512k,mode=755)
devpts on /dev/pts type devpts (rw,nosuid,noexec,relatime,seclabel,mode=600,ptmxmode=000)
debugfs on /sys/kernel/debug type debugfs (rw,noatime,seclabel)
/dev/sda1 on /mnt/sda1 type ufsd (rw,relatime,nls=utf8,uid=1023,gid=1023,fmask=3,dmask=2,force)
root@OpenWrt:~# cat /etc/config/nfs

root@OpenWrt:~#
root@OpenWrt:~# exportfs -a
root@OpenWrt:~#
root@OpenWrt:~# logread | tail -10
Wed Sep  2 10:26:56 2020 daemon.warn odhcpd[382]: DHCPV6 SOLICIT IA_NA from 0001000126a4b4f32c4d54e56f06 on eth0: no addresses available
Wed Sep  2 10:26:57 2020 daemon.warn odhcpd[382]: DHCPV6 SOLICIT IA_NA from 0001000126a4b4f32c4d54e56f06 on eth0: no addresses available
Wed Sep  2 10:26:57 2020 daemon.warn odhcpd[382]: DHCPV6 SOLICIT IA_NA from 0001000126a4b4f32c4d54e56f06 on eth0: no addresses available
Wed Sep  2 10:26:57 2020 daemon.warn odhcpd[382]: DHCPV6 SOLICIT IA_NA from 0001000126a4b4f32c4d54e56f06 on eth0: no addresses available
Wed Sep  2 10:26:57 2020 daemon.warn odhcpd[382]: DHCPV6 SOLICIT IA_NA from 0001000126a4b4f32c4d54e56f06 on eth0: no addresses available
Wed Sep  2 10:26:57 2020 daemon.warn odhcpd[382]: DHCPV6 SOLICIT IA_NA from 0001000126a4b4f32c4d54e56f06 on eth0: no addresses available
Wed Sep  2 10:26:57 2020 daemon.warn odhcpd[382]: DHCPV6 SOLICIT IA_NA from 0001000126a4b4f32c4d54e56f06 on eth0: no addresses available
Wed Sep  2 10:26:57 2020 daemon.warn odhcpd[382]: DHCPV6 SOLICIT IA_NA from 0001000126a4b4f32c4d54e56f06 on eth0: no addresses available
Wed Sep  2 10:30:21 2020 authpriv.info dropbear[7437]: Child connection from 10.0.0.9:50433
Wed Sep  2 10:30:21 2020 authpriv.notice dropbear[7437]: Auth succeeded with blank password for 'root' from 10.0.0.9:50433
root@OpenWrt:~#
root@OpenWrt:~# exportfs -s
/mnt/sda1  10.0.0.50(sync,wdelay,hide,no_subtree_check,sec=sys,ro,root_squash,all_squash)
root@OpenWrt:~#

boyu899, 2020-09-02:
NFS管理  共享目录  /mnt/sda1  允许的客户端*  选项rw,sync,root_squash,all_squash,insecure,no_subtree_check

启动后重启 执行命令
root@OpenWrt:~# ps
  PID USER       VSZ STAT COMMAND
    1 root      1592 S    /sbin/procd
   95 root      1140 S    /sbin/ubusd
  114 root       816 S    /sbin/xubus
  226 root      1188 S    /sbin/logd -S 64
  245 root      1716 S    /sbin/rpcd -s /var/run/ubus.sock -t 30
  265 root      1232 S    /sbin/mdadm --monitor --syslog --scan --config=/var/etc/mdadm.conf
  300 root       968 S    /usr/sbin/dropbear -F -P /var/run/dropbear.1.pid -p 22 -K 300 -T 3
  323 rpc       1260 S    /usr/sbin/rpcbind -f -w
  356 root      1680 S    /sbin/netifd
  382 root      1448 S    /usr/sbin/odhcpd
  508 root      2872 S    /usr/sbin/vsftpd -olisten_address=0.0.0.0 -olisten=YES -olisten_ipv6=NO /var/run/vsftpd/vsft
  540 root      1368 S    sh /lib/netifd/dns.script reload -i eth0
  554 root     20336 S    /usr/bin/minidlna -f /tmp/minidlna.conf
  581 root      1972 S    /usr/sbin/uhttpd -f -h /www -r OpenWrt -x /cgi-bin -t 60 -T 30 -k 20 -A 1 -n 3 -N 100 -R -p
  593 root      1536 S    /usr/sbin/dbus-daemon --system
  610 nobody    2384 S    avahi-daemon: running [OpenWrt.local]
  874 root      1196 S<   /usr/sbin/ntpd -n -N -S /usr/sbin/ntpd-hotplug -p 0.openwrt.pool.ntp.org -p 1.openwrt.pool.n
  927 root      2204 S    /usr/sbin/netatalk -d -F /etc/afp.conf
  957 nfs       1564 S    /usr/sbin/rpc.statd -p 32778 -o 32779 -F
  958 root      2056 S    /usr/sbin/rpc.mountd -p 32780 -F
  996 root      2632 S    /usr/sbin/afpd -d -F /etc/afp.conf
  997 root      1612 S    /usr/sbin/cnid_metad -d -F /etc/afp.conf
 1131 root      1196 S    lock /var/lock/dns_eth0.lock
 1132 root      1196 S    lock -w /var/lock/dns_eth0.lock
 1209 root       852 S    /usr/bin/wsdd2 -i eth0 -N OpenWrt -G WORKGROUP -b vendor:Realtek,model:Realtek_RTD1295,sku:F
 1292 root     33612 S    /usr/sbin/smbd -F
 1293 root     20112 S    /usr/sbin/nmbd -F
 1305 root     32320 S    {smbd-notifyd} /usr/sbin/smbd -F
 1306 root     32328 S    {cleanupd} /usr/sbin/smbd -F
 7428 root     34032 S    /usr/sbin/smbd -F
 7541 root      1092 R    /usr/sbin/dropbear -F -P /var/run/dropbear.1.pid -p 22 -K 300 -T 3
 7542 root      1204 S    -ash
 7555 root      1200 R    ps
root@OpenWrt:~# mount
/dev/mmcblk0p9 on /rom type squashfs (ro,relatime)
tmpfs on /rom/android type tmpfs (ro,noatime,seclabel,size=10240k)
proc on /proc type proc (rw,nosuid,nodev,noexec,noatime)
sysfs on /sys type sysfs (rw,nosuid,nodev,noexec,noatime,seclabel)
cgroup on /sys/fs/cgroup type cgroup (rw,nosuid,nodev,noexec,relatime,cpuset,cpu,cpuacct,blkio,memory,devices,freezer,net_cls,net_prio)
tmpfs on /tmp type tmpfs (rw,nosuid,nodev,noatime,seclabel)
/dev/loop0 on /overlay type ext4 (rw,relatime,seclabel,data=ordered)
overlayfs:/overlay on / type overlay (rw,noatime,lowerdir=/,upperdir=/overlay/upper,workdir=/overlay/work)
tmpfs on /dev type tmpfs (rw,nosuid,relatime,seclabel,size=512k,mode=755)
devpts on /dev/pts type devpts (rw,nosuid,noexec,relatime,seclabel,mode=600,ptmxmode=000)
debugfs on /sys/kernel/debug type debugfs (rw,noatime,seclabel)
/dev/sda1 on /mnt/sda1 type ufsd (rw,relatime,nls=utf8,uid=1023,gid=1023,fmask=3,dmask=2,force)
root@OpenWrt:~# cat /etc/config/nfs

config share
        option enabled '1'
        option path '/mnt/sda1'
        option clients '*'
        option options 'rw,sync,root_squash,all_squash,insecure,no_subtree_check'

root@OpenWrt:~# exportfs -a
root@OpenWrt:~# logread | tail -10
Wed Sep  2 10:34:42 2020 daemon.err uhttpd[581]: luci: accepted login on / for root from 10.0.0.9
Wed Sep  2 10:35:24 2020 daemon.warn odhcpd[382]: DHCPV6 SOLICIT IA_NA from 000100012392afad6ce85ce71400 on eth0: no addresses available
Wed Sep  2 10:37:40 2020 daemon.warn odhcpd[382]: DHCPV6 SOLICIT IA_NA from 000100012392afad6ce85ce71400 on eth0: no addresses available
Wed Sep  2 10:37:41 2020 daemon.warn odhcpd[382]: DHCPV6 SOLICIT IA_NA from 000100012392afad6ce85ce71400 on eth0: no addresses available
Wed Sep  2 10:37:43 2020 daemon.warn odhcpd[382]: DHCPV6 SOLICIT IA_NA from 000100012392afad6ce85ce71400 on eth0: no addresses available
Wed Sep  2 10:37:47 2020 daemon.warn odhcpd[382]: DHCPV6 SOLICIT IA_NA from 000100012392afad6ce85ce71400 on eth0: no addresses available
Wed Sep  2 10:37:52 2020 authpriv.info dropbear[7541]: Child connection from 10.0.0.9:50933
Wed Sep  2 10:37:52 2020 authpriv.notice dropbear[7541]: Auth succeeded with blank password for 'root' from 10.0.0.9:50933
Wed Sep  2 10:37:55 2020 daemon.warn odhcpd[382]: DHCPV6 SOLICIT IA_NA from 000100012392afad6ce85ce71400 on eth0: no addresses available
Wed Sep  2 10:38:12 2020 daemon.warn odhcpd[382]: DHCPV6 SOLICIT IA_NA from 000100012392afad6ce85ce71400 on eth0: no addresses available
root@OpenWrt:~# exportfs -s
/mnt/sda1  *(sync,wdelay,hide,no_subtree_check,anonuid=1023,anongid=1023,sec=sys,rw,root_squash,all_squash)
root@OpenWrt:~#

jjm2473, 2020-09-02:
看起来共享没有什么问题

jjm2473, 2020-09-02:
那可能是Z9X上挂载有问题了。扫描111端口是正常的，rpcbind就是监听111端口。

jjm2473, 2020-09-02:
你在其他设备挂载NFS试试吧，如果其他设备挂载成功，那就只能是Z9X的问题了

boyu899, 2020-09-02:
z9x 用kodi-18.7版本 也是空白 好的 我再试试其它设备 哦还有z9x已经出了smb版本 可以接硬盘盒远程了 不再需要x9s当nas了

jjm2473, 2020-09-02:
有个外国网友用的是x10，看起来x10的内核不支持nfsd，要想支持只能移植用户空间的nfs服务。https://github.com/jjm2473/rtd1295-openwrt18-porting/issues/22

x9s原始的openwrt支持nfs服务吗？

boyu899, 2020-09-02:
印象中x9s原始固件是没有nfs选项 z9s是支持的 官网本身有共享教程 基本就是打勾就可以共享了 我晚上试下原始固件能否开启nfs

boyu899, 2020-09-03:
zidoo原生x9s固件openwrt是没有nfs的 应该是本身就不支持

jjm2473, 2020-09-04:
嗯，我今天移植了一个用户空间的NFS服务，在我的机器上测试可以用，线上编译好给你试试

jjm2473, 2020-09-04:
更新了 https://github.com/jjm2473/openwrt/wiki/RTD129X-Openwrt-18.06

boyu899, 2020-09-04:
有问题 更新卡机 卡在安卓机器人画面 读条跑完

boyu899, 2020-09-04:
明天我试试原厂固件重新更新

jjm2473, 2020-09-04:
不用，试试这个方法，分分钟回滚到官方Openwrt https://github.com/jjm2473/openwrt/wiki/RTD129X-Openwrt-18.06#%E5%9B%9E%E6%BB%9A

boyu899, 2020-09-04:
回滚官方 原生官方 用最新那个固件 升级应该是失败的 卡在进度条跑完没反应 安卓机器人一直转圈

jjm2473, 2020-09-04:
如果不是把OTA包当成强刷包，那就不知道咋回事了。现在按照官方的强刷教程还能刷吗？

boyu899, 2020-09-05:
官方或你的9.04发布的X9S  重置或本地升级或强刷回原始固件再升级 都卡在安卓机器人 三种方法都试了 不折腾了 现在用上一个版本的 光荣退休了X9S Z9X已经挂了硬盘SMB共享影视

boyu899, 2020-09-05:
强刷貌似成功了 用回滚的方法 可以了

jjm2473, 2020-09-05:
回滚的方法不会显示机器人的

jjm2473, 2020-09-17:
X9S已支持NFS共享，升级到R20.09.04以上版本即可

closed 2022-10-08 by jjm2473