jjm2473-org/openwrta #2: 请问：TR的正确设置是什么？还有为什么每次重启后，种子都会提示你已经下载同一个种子，就红色感叹号

gujianwp, 2020-09-08:
另外是否可以加入QB，QB比TR要好用很多，期待您的回答。

gujianwp, 2020-09-08:
错误信息： | You already are downloading the same torrent. You may only leech from one location at a time. (2)
-- | --

如何更正？

gujianwp, 2020-09-08:
是否可以自己安装QB，软件目录里是没有的

jjm2473, 2020-09-09:
> 错误信息：	You already are downloading the same torrent. You may only leech from one location at a time. (2)
> 如何更正？

TR本来就是这样的，重启就会导致clientid变化，tracker会认为换机器了，这个一般半小时以后就会消失了。除非你经常重启，不然不需要在意这个。我有时间也会看看tr的代码看能不能保存clientid。

> 是否可以自己安装QB，软件目录里是没有的

OP官方的仓库就没有QB，我看能不能从其他仓库拷贝吧。TR对我来说够用了。

jjm2473, 2020-09-09:
> TR的正确设置是什么？

默认设置有啥问题吗？

贴一个我用的配置（/etc/config/transmission）：
```
config transmission 'config'
	option config_overwrite '1'
	option group 'media_rw'
	option mem_percentage '30'
	option nice '10'
	option alt_speed_time_day '127'
	option bind_address_ipv4 '0.0.0.0'
	option bind_address_ipv6 '::'
	option blocklist_enabled 'false'
	option dht_enabled 'true'
	option download_queue_enabled 'true'
	option download_queue_size '4'
	option encryption '1'
	option idle_seeding_limit_enabled 'false'
	option incomplete_dir_enabled 'false'
	option lazy_bitfield_enabled 'true'
	option message_level '1'
	option peer_limit_global '240'
	option peer_limit_per_torrent '60'
	option peer_port_random_on_start 'false'
	option peer_socket_tos 'default'
	option pex_enabled 'true'
	option port_forwarding_enabled 'true'
	option preallocation '1'
	option queue_stalled_enabled 'true'
	option queue_stalled_minutes '30'
	option ratio_limit_enabled 'false'
	option rpc_authentication_required 'false'
	option rpc_bind_address '0.0.0.0'
	option rpc_enabled 'true'
	option rpc_host_whitelist '127.0.0.1,192.168.1.*'
	option rpc_host_whitelist_enabled 'false'
	option rpc_port '9091'
	option rpc_url '/transmission/'
	option rpc_whitelist_enabled 'false'
	option scrape_paused_torrents_enabled 'true'
	option script_torrent_done_enabled 'false'
	option seed_queue_enabled 'false'
	option trash_original_torrent_files 'false'
	option umask '18'
	option upload_slots_per_torrent '14'
	option utp_enabled 'true'
	option scrape_paused_torrents 'true'
	option watch_dir_enabled 'false'
	option config_dir '/mnt/sataa2/transmission'
	option cache_size_mb '100'
	option alt_speed_enabled 'true'
	option alt_speed_down '5000'
	option alt_speed_up '2000'
	option speed_limit_down_enabled 'true'
	option speed_limit_down '3096'
	option speed_limit_up_enabled 'true'
	option speed_limit_up '1024'
	option download_dir '/mnt/sataa2/Download'
	option rename_partial_files 'false'
	option start_added_torrents 'false'
	option lpd_enabled 'true'
	option alt_speed_time_enabled 'true'
	option alt_speed_time_begin '60'
	option alt_speed_time_end '360'
	option peer_port '51234'
	option user 'nobody'
	option enabled '1'
```

注意这个配置会每晚1点到6点把限速调高的，其他就自己看吧。

gujianwp, 2020-09-09:
老板好，你这个缓存设置的是多大啊？我们小睿4g内存

jjm2473, 2020-09-09:
100M

jjm2473, 2020-09-10:
软件仓库中增加了qBittorrent，如果需要安装qBittorrent，可以升级到[这个固件](https://github.com/jjm2473/openwrt/wiki/RTD129X-Openwrt-18.06#openwrta-1806-r200910)，如果不想升级，也可以添加这个opkg源`https://raw.githubusercontent.com/jjm2473/cdn/master/openwrt/release/18.06.8/targets/realtek/rtd129x/packages`

然后在后台的软件包页面搜索安装`luci-i18n-qbittorrent-zh-cn`即可。

chenyu3131, 2020-09-24:
QB插件怎么找不到，已经刷最新固件了，能不能贴个图出来😜

jjm2473, 2020-09-24:
执行`opkg update`看看输出是什么

jjm2473, 2020-09-24:
[image]

chenyu3131, 2020-09-24:
我用winscp执行命令的，看不到qb插件，ssh这个软件下载下来设置中文失败，不会使用，是不是只能用ssh执行命令才行？进入盒子openwrt后台刷新，然后用您的软件包名称搜索，也是失败的！大佬出个简单教程，万分感谢！芝杜机顶盒官方群也没人会搞！------------------&nbsp;原始邮件&nbsp;------------------
发件人:&nbsp;&quot;练亮斌&quot;<notifications@github.com&gt;
发送时间:&nbsp;2020年9月24日(星期四) 晚上8:42
收件人:&nbsp;&quot;jjm2473/openwrt&quot;<openwrt@noreply.github.com&gt;;
抄送:&nbsp;&quot;chenyu3131&quot;<598677976@qq.com&gt;;&quot;Comment&quot;<comment@noreply.github.com&gt;;
主题:&nbsp;Re: [jjm2473/openwrt] [已解决] Transmission的正确设置是什么？还有为什么每次重启后，都会提示你已经下载同一个种子，就红色感叹号？如何安装qBittorrent？ (#2)

jjm2473, 2020-09-24:
我没用过winscp，据我所知它是文件传输工具，不知道能不能用它来ssh。

既然你能执行命令，麻烦执行`opkg update`，把输出贴出来。

如果你之前手动改过软件源，那你的设置会覆盖新固件的设置，需要给软件源加上`src/gz jjm_core https://raw.githubusercontent.com/jjm2473/cdn/master/openwrt/release/18.06.8/targets/realtek/rtd129x/packages`。（如果你会改软件源，那自然知道[我前面说的](https://github.com/jjm2473/openwrt/issues/2#issuecomment-690235988)是什么）

如果不知道怎么看软件源，那就还是执行`opkg update`再说吧。

一些简单的工具，例如ssh，我不可能自己写教程的，毕竟到处都能找到教程，我只会提供这个固件或者这个设备特定的教程。要是连所有简单的东西都要我来写教程，那我就不用做实事了。

chenyu3131, 2020-09-27:
[image]
感谢大佬不厌其烦，帮忙看看刷出来的固件，软件包配置有没有错误？点击刷新，然后搜索QB，搜索不到！再次感谢😊

chenyu3131, 2020-09-27:
opkg update
[image]

chenyu3131, 2020-09-27:
[image]

chenyu3131, 2020-09-27:
[image]

jjm2473, 2020-09-27:
软件源没问题。不过看不到后面的图片，不知道你的输出是什么。

点击刷新软件列表有什么报错吗？如果刷新超时，还是用命令刷吧，方便看输出。

susie1986, 2022-02-08:
1.qBittorrent安装上web ui 会失效 用ssh连上x9s 用/usr/bin/qbittorrent-nox 能打开web ui 但是过一会就又不行了，有点难搞。
2.Transmission 下载会出现Unable to save resume file: Permission denied  恢复初始设置都无法解决，硬盘也格式化了，最后root了x9s 用chmod -R 0777 下载目录 解决这个问题。。。。折腾了4天。nobody这个是什么权限？