jjm2473-org/openwrta #3: x9s刷了OpenwRTA 18.06 R20.09.30版本，openwrt启动不了

Hajimu007, 2020-10-03:
开机Android功能正常，但openwrt似乎没有启动，本机“设置”--“更多”--“openwrt设置”，所有的功能开关都是关闭的，打开就提示“fail”，远程登陆被拒绝。刷了两次都是一样。请问是啥问题啊？

jjm2473, 2020-10-05:
听起来好像是openwrt启动失败了，如果你会用adb的话，进安卓系统看下进程列表。如果不会的话，我建议直接试试刷重置包。

一般情况下OP会使用磁盘的空闲空间作为overlay分区，但是如果没有空闲空间就会使用以前官方的overlay分区，可能导致OP启动失败，这种情况可以试试刷重置包，具体看这里 https://github.com/jjm2473/rtd1295-openwrt18-porting#%E4%B8%89openwrt%E9%85%8D%E7%BD%AE%E9%87%8D%E7%BD%AE%E5%8C%85

jjm2473, 2020-10-05:
如果你能使用ttl的话，可以把启动日志发出来看看

Hajimu007, 2020-10-05:
我重刷了官方版本，有时间再折腾

Hajimu007, 2020-10-06:
OpenWrt 19.07.0-rc2 r10775-db8345d8e4 / LuCI openwrt-19.07 branch git-19.365.55415-a44e6b3（x9s_op18_2.0.0.signed）
这个版本可以运行，这个版本是不是不能安装qBittorrent？添加了软件源还是找不到。

jjm2473, 2020-10-07:
试试增加这些软件源
```
src/gz jjm_core https://raw.githubusercontent.com/jjm2473/cdn/master/openwrt/release/18.06.8/targets/realtek/rtd129x/packages
src/gz jjm_base https://raw.githubusercontent.com/jjm2473/cdn/master/openwrt/release/18.06.8/packages/aarch64_cortex-a53/base
src/gz jjm_luci https://raw.githubusercontent.com/jjm2473/cdn/master/openwrt/release/18.06.8/packages/aarch64_cortex-a53/luci
src/gz jjm_packages https://raw.githubusercontent.com/jjm2473/cdn/master/openwrt/release/18.06.8/packages/aarch64_cortex-a53/packages
```
然后关掉opkg的签名校验，也就是删掉`option check_signature`，
然后更新软件列表，安装qBittorrent。
不知道能不能用，通常可以。

jjm2473, 2021-03-21:
@Hajimu007 新固件修复了这个问题 https://github.com/jjm2473/openwrt/wiki/RTD129X-Openwrt-18.06#openwrta-1806-r210320

closed 2021-03-21 by jjm2473

Hajimu007, 2021-04-13:
很久没有折腾了，再试一试，谢谢