jjm2473-org/openwrta #4: 请教运行双系统的原理？

jadeCheng2017, 2020-10-10:
你好，同时运行Android和OpenWRT的原理大致是什么样子的？
两者共用kernel，移植了OpenWRT上的相关应用（luCI、NAS、dlna等等）到Android系统吗？

jjm2473, 2020-10-10:
两者共用kernel，但不是把应用移植到Android，而是利用linux的命名空间对一些资源进行隔离，然后原始的1号进程启动Android的init，fork出一个进程启动OpenWRT的init。具体的代码可以看这里 https://github.com/jjm2473/openwrt/tree/openwrt-18.06/package/boot/rtk-init/src

jjm2473, 2020-10-10:
由于代码是逆向出来的，有些变量没改名称，所以不太好看，不过程序比较简单，原理也清楚，我也就没去整理了

jadeCheng2017, 2020-10-10:
> 两者共用kernel，但不是把应用移植到Android，而是利用linux的命名空间对一些资源进行隔离，然后1号进程（/etc/init）fork出一个进程启动OpenWRT的init，原始的1号进程启动Android的init。
> 所以总共涉及到三个init程序，其中/etc/init是最早启动的进程，也是它启动两个系统的，具体的代码可以看这里 https://github.com/jjm2473/openwrt/tree/openwrt-18.06/package/boot/rtk-init/src

非常感谢！我先学习一下，有问题在请教~~~

closed 2020-10-10 by jadeCheng2017