jjm2473-org/openwrta #5: 无法下载transmission-2.94-streaming导致编译失败问题

jadeCheng2017, 2020-10-12:
编译openRTA时，提示下载https://sources.openwrt.org/transmission-2.94-streaming.tar.xz失败，但是存在https://sources.openwrt.org/transmission-2.94.tar.xz ，手动下载后，更名为transmission-2.94-streaming.tar.xz，放入本地openRTA的dl目录，编译仍然失败，提示：
`. /home/jade/work/code/OpenwRTA/openwrt/include/shell.sh; xzcat /home/jade/work/code/OpenwRTA/openwrt/dl/transmission-2.94-streaming.tar.xz | tar -C /home/jade/work/code/OpenwRTA/openwrt/build_dir/target-aarch64_cortex-a53_musl/transmission-2.94-streaming/.. -xf - 
[ ! -d ./src/ ] || cp -fpR ./src/. /home/jade/work/code/OpenwRTA/openwrt/build_dir/target-aarch64_cortex-a53_musl/transmission-2.94-streaming

Applying ./patches/010-fix-port-test.patch using plaintext: 
can't find file to patch at input line 19
Perhaps you used the wrong -p or --strip option?
The text leading up to this was:
--------------------------
|From d6655cca7db1b960456811b8206ce222700e010d Mon Sep 17 00:00:00 2001
|From: Rosen Penev <rosenp@gmail.com>
|Date: Tue, 30 Jan 2018 13:19:58 -0800
|Subject: [PATCH] Remove compressed responses from web servers.
|
|While zlib is mandatory for transmission, it is not mandatory for curl.
|
|A libcurl that has been compiled with no support for zlib will return no data if compressed responses are set to on.
|
|Signed-off-by: Rosen Penev <rosenp@gmail.com>
|---
| libtransmission/web.c | 2 +-
| 1 file changed, 1 insertion(+), 1 deletion(-)
|
|diff --git a/libtransmission/web.c b/libtransmission/web.c
|index c7f0627..db34976 100644
|--- a/libtransmission/web.c
|+++ b/libtransmission/web.c
--------------------------
No file to patch.  Skipping patch.
1 out of 1 hunk ignored
Patch failed!  Please fix ./patches/010-fix-port-test.patch!
Makefile:179: recipe for target '/home/jade/work/code/OpenwRTA/openwrt/build_dir/target-aarch64_cortex-a53_musl/transmission-2.94-streaming/.prepared_a1aa88dfd81dc0b0858ddbec1e094a22_6664517399ebbbc92a37c5bb081b5c53' failed`

是否需要到openwrt项目开单吗？

jadeCheng2017, 2020-10-12:
非问题，关单。

closed 2020-10-12 by jadeCheng2017

jjm2473, 2020-10-12:
这个transmission不是官方的，而是我改过的，项目在 https://github.com/jjm2473/transmission-streaming， 所以openwrt肯定没缓存源代码包。错误发生在更早之前，也就是从原始地址下载失败了，手动下载可以到项目的Releases页面。
如果想使用官方的transmission，删除 .github/workflows/overwrite/feeds/packages/net/transmission 目录，然后进入 feeds/packages 执行 `git checkout net/transmission` 即可。