jjm2473-org/openwrta #7: 小睿私人云刷X9S，刷了openwrt，安卓通过wifi连接的话，openwrt连不上外网

fengshutao, 2020-11-25:
首先非常感谢jjm2473大佬的新版openwrt，我是小睿私人云刷的X9S，同时内置了J大的这个新版OP，但是现在我就遇到这个问题，因为是当机顶盒+PT下载用，一开始并没有用有线，而是安卓里面连接了wifi，折腾了几天tr始终无法下载，最终测试发现openwrt并没有连上外网，后来换有线连接后，问题就没有了，可以正常下载。安卓和op感觉是完全两个独立不相干的系统，不知道J大能不能解决这个问题，我并不想连接有线，就想连接wifi做机顶盒+PT下载。

jjm2473, 2020-11-26:
wifi应该可以上网的，但是不能使用静态ip，需要dhcp。静态ip的话，安卓那边不会通知op更新dns，这是zidoo的bug，我给他们反馈过。

jjm2473, 2020-11-26:
安卓和op并不是完全独立的，网络栈只有一套。zidoo的固件是安卓控制网络连接，op不能修改网络配置，改了可能导致不能上网，严重情况下系统会无限重启。

fengshutao, 2020-11-26:
非常感谢你能回复，我直接连接wifi，没有用静态ip，但是op一样连不上外网，软件更新opkg update都更新不了，有线连接就可以。你的意思是反正不能在op里面设置网络连接方式，是吧？另外反馈个问题就是，你的源好像都连不上，下面几个大学的源可以。------------------&nbsp;原始邮件&nbsp;------------------
发件人:&nbsp;&quot;练亮斌&quot;<notifications@github.com&gt;
发送时间:&nbsp;2020年11月26日(星期四) 晚上6:21
收件人:&nbsp;&quot;jjm2473/openwrt&quot;<openwrt@noreply.github.com&gt;;
抄送:&nbsp;&quot;fengshutao&quot;<1985352381@qq.com&gt;;&quot;Author&quot;<author@noreply.github.com&gt;;
主题:&nbsp;Re: [jjm2473/openwrt] 小睿私人云刷X9S，刷了openwrt，安卓通过wifi连接的话，openwrt连不上外网 (#7)

jjm2473, 2020-11-27:
我的源是在github上的，电信的DNS会劫持github的域名，导致无法连接，把路由器的dns改成114就能更新了。

不能在op里面改网络设置，你改过，那先恢复默认配置试试。如果还有问题，那贴一下固件版本（包括android和op），还有ping qq.com的输出。

还有在刷我的op之前，在X9S自带的op可以正常联网吗？如果之前就不行，我怀疑这个X9S的固件可能还改了什么东西，导致wifi连接时不通知op更新dns。

niubility000, 2021-01-03:
我也是用的小睿私人云刷的芝杜+openwrt，用起来非常流畅。我想拿openwrt搭个人云。发现每次重启,mac地址都会变啊？？？这可怎么绑定ip呢？？？

fengshutao, 2021-01-04:
没遇到这个问题，我的mac不会变

jjm2473, 2021-01-05:
> 我也是用的小睿私人云刷的芝杜+openwrt，用起来非常流畅。我想拿openwrt搭个人云。发现每次重启,mac地址都会变啊？？？这可怎么绑定ip呢？？？

MAC变化跟我的OP没有关系，就算你不刷我的OP也一样会每次启动改变MAC。

小睿私人云刷机以后需要在UART里面修改MAC地址，不然每次启动都会生成MAC。
具体操作是在UART里面执行`uart_write [MAC地址]`（把`[MAC地址]`换成你希望的MAC地址）。
小睿私人云刷机教程应该有说明这个问题，仔细看下教程好吧。

niubility000, 2021-01-05:
> > 我也是用的小睿私人云刷的芝杜+openwrt，用起来非常流畅。我想拿openwrt搭个人云。发现每次重启,mac地址都会变啊？？？这可怎么绑定ip呢？？？
> 
> MAC变化跟我的OP没有关系，就算你不刷我的OP也一样会每次启动改变MAC。
> 
> 小睿私人云刷机以后需要在UART里面修改MAC地址，不然每次启动都会生成MAC。
> 具体操作是在UART里面执行`uart_write [MAC地址]`（把`[MAC地址]`换成你希望的MAC地址）。
> 小睿私人云刷机教程应该有说明这个问题，仔细看下教程好吧。

谢谢。搞定了。恩山那个教程没写要绑定mac，张大妈那里的文章倒是写了。亏得看了这里，直接就改好了。。

jjm2473, 2021-06-27:
@fengshutao ZIDOO的系统有bug，用wifi的时候不会通知openwrt刷新DNS，安装这个APK可以解决 https://github.com/jjm2473/RealtekOpenwrtDNS/releases

jjm2473, 2021-06-27:
汇总到 #20

closed 2021-06-27 by jjm2473