jjm2473-org/openwrta #9: 【公告】关于软件仓库无法访问的问题

jjm2473, 2021-01-06:
由于国内的运营商劫持了DNS，`raw.githubusercontent.com`会被解析到错误的IP地址，导致我的软件仓库无法访问。
前段时间还可以通过把DNS改成114.114.114.114来解析到正确的IP地址，但是最近这个方法也行不通了，运营商已经完全劫持了DNS请求。

目前最好的方式还是科学上网来防止网络劫持。

如果没有条件科学上网，那么想办法找到`raw.githubusercontent.com`的IP地址，然后把域名和IP地址在hosts里面绑定，这样可能就能访问了，但是某些IP地址也可能已经被墙，所以这个方法很麻烦，需要不断尝试各个IP，不推荐使用。

abcwers, 2021-01-17:
I found, that github.com ip adress 140.82.113.4 , and it can redirect to raw.githubusercontent.com .
	
	links examples:
	
	https://raw.githubusercontent.com/jjm2473/rtd1295-blob/raw/openwrt-cc-packages/packages
	https://github.com/jjm2473/rtd1295-blob/raw/openwrt-cc-packages/packages
	https://140.82.113.4/jjm2473/rtd1295-blob/raw/openwrt-cc-packages/packages
	
	
	https://raw.githubusercontent.com/jjm2473/rtd1295-blob/openwrt-cc-packages/packages/Packages
	https://github.com/jjm2473/rtd1295-blob/raw/openwrt-cc-packages/packages/Packages
	https://140.82.113.4/jjm2473/rtd1295-blob/raw/openwrt-cc-packages/packages/Packages

jjm2473, 2021-01-17:
> ```
> I found, that github.com ip adress 140.82.113.4 , and it can redirect to raw.githubusercontent.com .
>
> links examples: [...]

Thanks, but this IP may outdated some day. There is a DNS MITM attack from ISP, we don't have good solution for everyone, just VPN works.

leenchan, 2021-03-18:
"除了这些，由于Android系统同时在运行，Android系统定义的部分权限会影响到Openwrt，Android是通过GID（准确来说是supplementary group IDs）来管理权限的，这也就导致了在RtkWRT中部分软件必须以某个特定UID或GID运行才正常。OpenwRTA修改了libc以自动赋予进程网络访问权限和外部存储设备读取权限，不必再纠结UID和GID的选择。（如果某进程绕过libc进行系统调用，丢弃自己的权限，那我也没办法）"

你好，大神，想请教一下你怎么修改 libc  以自动赋予进程网络访问权限和外部储存设备读取权限

jjm2473, 2021-03-18:
@leenchan  我用的是musl libc，所以我给musl打了补丁，可以看看这个
https://github.com/jjm2473/openwrt/blob/openwrt-18.06/toolchain/musl/patches/999-rtk_android_permission.patch

leenchan, 2021-03-19:
> @leenchan 我用的是musl libc，所以我给musl打了补丁，可以看看这个
> https://github.com/jjm2473/openwrt/blob/openwrt-18.06/toolchain/musl/patches/999-rtk_android_permission.patch

好的，谢谢你，大神，我去研究一下先~

susie1986, 2022-02-06:
问题解决了，无法更新是没有装Realtek Openwrt Wifi DNS修复器 安装上用科学工具软件仓就连上了

jjm2473, 2022-02-08:
哈哈，原来是这个问题

susie1986, 2022-02-08:
是的，这个问题折腾了一个星期 。。