jjm2473-org/openwrta #10: z9s刷入后，接了个硬盘想当私人云盘用，结果能创建目录不能写入文件

qq5274202, 2021-01-08:
您好，z9s刷入后，接了个硬盘想当私人云盘用，结果能创建目录不能写入文件

硬盘格式为ntfs，使用nfs协议，把z9s当私人云盘，用于备份手机资料到z9s的硬盘上

硬盘接上后，openwrt识别硬盘/dev/sataa1，将/dev/sataa1挂载为/mnt/sda1，挂载格式自动识别是vfat（我找不到ntfs选项），磁盘大小，已使用空间都识别出来了

我用ssh进入盒子，chmod 0777 /mnt/sda1，依然无法写入文件，能创建目录

试了chmod 0777 /dev/sataa1，也不行

jjm2473, 2021-01-08:
NFS的权限控制比较复杂，不只是服务端需要对应权限，客户端也需要登录相应的用户才行，如果能用Samba尽量用Samba，没这么多麻烦。

如果确实要用NFS，那么把`/etc/config/nfs`文件的内容发出来看看，以及客户端挂载命令也发出来看看。

jjm2473, 2021-01-08:
ntfs的挂载问题，其实是因为这个op使用的是ufsd驱动，并没有NTFS的单独驱动，所以选择ufsd就行了，exfat也是使用ufsd。如果是自动识别的，应该是ntfs或者ufsd才对。

jjm2473, 2021-01-08:
先尝试把磁盘挂载格式改为ufsd吧，然后重启一下，看看nfs是不是正常

qq5274202, 2021-01-08:
好的，感谢，晚上回去了我把vfat改为ufsd试试，客户端是手机上装的一个app，万由nas的u-file这个工具，不是用命令挂载的，手机用samba连盒子连不上，能发现盒子的samba，但无法连接，但是用电脑windows10，能通过samba连接并新增删除

jjm2473, 2021-01-08:
客户端的话，不知道支持的NFS版本是多少。连不上samba可能是因为手机的samba协议太老了，华为手机的samba就是鉴权协议太老，新的samba4已经不支持旧的鉴权。

qq5274202, 2021-01-08:
感谢,磁盘挂载格式改为ufsd了,但问题依旧
仔细看了下nfs,默认的挂载参数rw,sync,root_squash,all_squash,insecure,no_subtree_check
其中root_squash,all_squash不应该同时出现,于是删除all_squash,同时将root_squash改为no_root_squash(我家的网络是启用了mac黑白名单,也不曝露在外网)
最终为:rw,sync,no_root_squash,insecure,no_subtree_check
问题解决
再次感谢

关于samba,这个u-file应该支持到了最新samba v3,因为我用u-file连接家里的群晖速度是能达到35m的,也支持大文件

qq5274202, 2021-01-08:
nfs只能匿名登陆,根本也不存在安全性,干脆no_root_squash省事省心

jjm2473, 2021-01-09:
NFS可以鉴权的，可能你的客户端不支持吧。

我说的samba协议不是指传输协议，是其中的鉴权协议，新的samba4，应该是4.11以后不支持NTLM V1的鉴权了，有些客户端又只有这个鉴权协议。

qq5274202, 2021-01-11:
恩，软件在连接nfs的时候没有提供帐户密码输入窗口，是客户端不支持，安卓上能够用nfs协议的客户端软件不多，能够指定备份文件夹的并自动备份的软件暂时还没发现，就这个u-file能达成功能需求，nfs权限配置只在Nas中见到过，openwrt里好像都是命令添加
samba估计也是客户端软件层面问题，因为我用windows连接能正常使用
感谢您这个openwrt修改版，再想了解一下，您用z9s挂载bt吗，我担心这盒子的散热可能不适合长时间挂机

jjm2473, 2021-01-11:
挂pt没啥问题的，我都是挂种子从不关机的，z9s功耗5w（2.5寸硬盘接SATA接口，不是USB，没试过USB连接硬盘），非常适合挂机。不过需要注意，如果种子非常多，体积太大的时候，注意系统会不会变得不稳定。我比较常用transmission，同时软件仓库也提供了qBittorrent。如果你家网络有公网IP，可以下限制同时连接数和速率。

NFS的权限估计就是按照openwrt里面的linux用户密码来的，不过openwrt默认没有用户管理的命令（useradd等），比较麻烦。还有一个叫UAM的插件式鉴权模块，不知道NFS支不支持。

closed 2021-01-14 by qq5274202