jjm2473-org/openwrta #11: x9s刷机后刷OpenwRTA 18.06 R20.12.28 也无法启动

nmsbnmsb1, 2021-01-17:
系统更新到 2.1.46，第一次刷完 OpenwRTA 18.06 R20.12.28 是ok的，一切都正常。安装完qBittorrent，改了一下下载临时目录后，整个机子就挂了。
重启后android能进入，但是openwrt无法启动。
重置配置后依旧无法启动。
强制刷机再刷入OpenwRTA 18.06 R20.12.28，也是无法启动。而且插入的硬盘，文件管理器里看不到。
现在只能先回滚到官方版本，但是我需要安装qBittorrent，不知道怎么解决呀 T.T

jjm2473, 2021-01-17:
我更新了重置OpenwRTA的教程，适用于无法在Android设置里面重置的情况：
https://github.com/jjm2473/openwrt/wiki/RTD129X-Openwrt-18.06#%E9%87%8D%E7%BD%AEopenwrta

nmsbnmsb1, 2021-01-18:
感谢回复，刚试了下重置也无效。
然后只有 Release x9s 2.0.0 with bundle(X9S 2.0.0完全版) 这个版本可以，看来和 issue#3 的兄弟一样的问题>_<
这个用adb怎么调试呢？

jjm2473, 2021-01-19:
刷了重置包还是不能启动吗？

adb不知道能不能看到开机日志，试试开机以后尽快执行 `adb logcat`，看看最前面几百行是什么。

还有可以尝试刷 ***OpenwRTA 18.06 R20.08.29*** 看有没有问题。

jjm2473, 2021-01-19:
还有确认下系统版本是不是2.1.46，官方最新的是2.1.42

nmsbnmsb1, 2021-01-19:
嗯，是2.1.42，刷了 OpenwRTA 18.06 R20.08.29 还是无法启动。。。然后没有usb线做调试 -。-
只能先换回官方的凑合用
感谢解答

jjm2473, 2021-01-19:
adb 不需要数据线，通过网络连接就行

tommytianhm, 2021-03-19:
我的症状跟这个一模一样，第一次刷成功了，然后改了qb的配置，然后崩溃，重启后再也不能进后台了，USB挂硬盘也不识别了。看来这个现象是可以重现的，希望JJm2473能抽空解决，谢谢。

jjm2473, 2021-03-20:
OK，不过我没有X9S，只有Z9S，目前没看到反馈Z9S出问题

jjm2473, 2021-03-21:
@nmsbnmsb1 修复了 https://github.com/jjm2473/openwrt/wiki/RTD129X-Openwrt-18.06#openwrta-1806-r210320

tommytianhm, 2021-03-21:
@jjm2473 刚才刷了最新的固件，问题依旧：无法进后台，无法识别USB硬盘。

jjm2473, 2021-03-21:
@tommytianhm 刷完以后有刷重置包吗？

tommytianhm, 2021-03-21:
@jjm2473果然，重置之后就可以了，谢谢了

nmsbnmsb1, 2021-03-21:
@jjm2473 辛苦，应该没问题了。感谢

closed 2021-06-27 by jjm2473