jjm2473-org/openwrta #12: 请问下Z9S如何修改default.prop使其重启后不会还原

billqq1, 2021-01-28:
想修改default.prop 文件最后persist.sys.usb.config=“” ，每次重启都要打开usb调试模式很麻烦。

jjm2473, 2021-01-29:
如果是自己编译：

1. 克隆 https://github.com/jjm2473/realtek-android-rootfs ，检出对应版本的分支（Z9S应该用7.1.1分支）
2. 你要修改的文件是 android/default.prop
3. 执行 `./build.sh` 打包，生成 android-7.1.1.tar.gz
4. 克隆这个项目，并修改 package/system/rtk-android-rootfs/Makefile，把 PKG_SOURCE 改成 android-7.1.1.tar.gz，把 PKG_SOURCE_URL 的值改成上一步打包的文件父目录（/结尾），PKG_HASH 的值改成 android-7.1.1.tar.gz 文件的sha256值，然后编译固件就行了。编译过程参考 https://github.com/jjm2473/openwrt/wiki

如果不想编译：
1. 电脑上需要Linux或者macOS（Windows不知道行不行），并安装JAVA，unsquashfs，mksquashfs
1. 下载固件（zip包），将固件里面的squashfs1.img解压出来
3. 用unsquashfs工具解压squashfs1.img，默认会输出到squashfs-root
4. 你要修改的文件在 squashfs-root/mnt/android/default.prop
5. 改好以后，用mksquashfs打包squashfs-root，打包以后的文件名为squashfs1.img，具体的命令是 `mksquashfs squashfs-root squashfs1.img -comp xz -all-root`
6. 克隆或者下载这个项目，将上一步生成的 squashfs1.img 复制到 target/linux/realtek/image/android，并 cd 到target/linux/realtek/image/android
7. 执行 `./mkota.sh squashfs1.img`，将会生成OTA.zip，这就是可以刷机的包。

billqq1, 2021-01-30:
太感谢了，回头试一下~

jjm2473, 2021-01-31:
不用这么麻烦了，而且还不一定有效，因为Realtek改了adb设置的逻辑。

还是下载这个apk吧，可以直接安装 
https://github.com/jjm2473/RealtekAdbdAuto/releases/tag/v1.0.0

jjm2473, 2021-01-31:
直接在app里面开启调试，不用再进入系统设置里面改了

billqq1, 2021-01-31:
下了,超级好用哈.重启机子,电脑adb直接可连,解决了研究了两周的问题.
能否再请教下,我最终目的是利用起Z9S的无线AP功能,并实现定时开关AP.
原以为通过shell,基本的linux指令即可控制wifi ap的控制,但经adb logcat后发现开关ap是由一系列指令完成,放弃此路后,想通过模拟按键实现input keyevent 23.
能否有更好的方案来实现上述功能呢?

billqq1, 2021-01-31:
目前已可实现不开电视,通过电脑批处理文件来模拟遥控器打开AP功能.但如果能脱离电脑完全实现自动话就更完美了. 

附上批处理bat,
D:\tools\adb\adb connect 192.168.31.117
TIMEOUT /T 5
D:\tools\adb\adb shell input keyevent 183
TIMEOUT /T 1
D:\tools\adb\adb shell input keyevent 22
TIMEOUT /T 1
D:\tools\adb\adb shell input keyevent 22
TIMEOUT /T 1
D:\tools\adb\adb shell input keyevent 22
TIMEOUT /T 1
D:\tools\adb\adb shell input keyevent 20
TIMEOUT /T 1
D:\tools\adb\adb shell input keyevent 20
TIMEOUT /T 1
D:\tools\adb\adb shell input keyevent 20
TIMEOUT /T 1
D:\tools\adb\adb shell input keyevent 20
TIMEOUT /T 1
D:\tools\adb\adb shell input keyevent 22
TIMEOUT /T 1
D:\tools\adb\adb shell input keyevent 23

或能否通过类似crontab的功能定时执行模拟按键的shell脚本,最终实现目标.请指教.

jjm2473, 2021-01-31:
我自己没怎么研究过自动化。你看有没有按键精灵这一类的软件吧，我以前用过一个叫Automate的工具，不知道有没有开启AP的功能。
更高级的玩法可以参考这个 https://hyb1996.github.io/AutoJs-Docs/#/

jjm2473, 2021-01-31:
Z9S的wifi模块稳定性不太好吧，开AP好用吗

billqq1, 2021-01-31:
好的.我也是用来对主AP覆盖不到的地方的补充,目前看还行,也就上个网页,刷个抖音之类,也不游戏,下载.

jjm2473, 2021-01-31:
感觉买个矿渣路由器更方便，还省电

billqq1, 2021-01-31:
我也就物尽其用,能用就用呗,就不另外添至设备了,家里设备多了乱~~
Z9S也省电啊,我测了下开AP待机只有14 5W,挂硬盘也就20来W.

closed 2021-02-14 by jjm2473