jjm2473-org/openwrta #13: 实在是连不上软件仓库，下不到qbittorrent

beizi2, 2021-02-22:
不知道能不能直接提供个qbittorrent的ipk文件？非常感谢

jjm2473, 2021-02-23:
现在要科学上网才能连上github的软件仓库。

如果要手动安装ipk，在这里下载 https://github.com/jjm2473/cdn/tree/master/openwrt/release/18.06.9/packages/aarch64_cortex-a53/base 
可能需要安装一些依赖库，都能在 https://github.com/jjm2473/cdn/tree/master/openwrt/release/18.06.9/packages/aarch64_cortex-a53 下面找到

jjm2473, 2021-03-21:
@WangLeiCloudcade https://github.com/ 和 https://raw.githubusercontent.com/ 的功能是不一样的，不能直接替换。
国内可以访问github.com，但是不能访问raw.githubusercontent.com。
不过你说加上/tree就能用，如果确实可行，我之后就把源替换成这个

jjm2473, 2021-03-21:
@WangLeiCloudcade 你能把替换好的网址发一下吗，我看能不能访问。（因为github.com通常只能访问页面，不能访问原始文件）

WangLeiCloudcade, 2021-03-23:
```
#src/gz jjm_core https://github.com/jjm2473/cdn/tree/master/openwrt/release/18.06.9/targets/realtek/rtd129x/packages
#src/gz jjm_base https://github.com/jjm2473/cdn/tree/master/openwrt/release/18.06.9/packages/aarch64_cortex-a53/base
#src/gz jjm_luci https://github.com/jjm2473/cdn/tree/master/openwrt/release/18.06.9/packages/aarch64_cortex-a53/luci
#src/gz jjm_packages https://github.com/jjm2473/cdn/tree/master/openwrt/release/18.06.9/packages/aarch64_cortex-a53/packages
```

这是我自己实际在用的源地址，不过为了方便快速链接，我把上面源的内容全部下到本地，做了ftp的方式来连接了。实测应该是没问题可以下载到ipk的

```
src/gz jjm_core ftp://192.168.2.101/jjm_core
src/gz jjm_base ftp://192.168.2.101/jjm_base
src/gz jjm_luci ftp://192.168.2.101/jjm_luci
src/gz jjm_packages ftp://192.168.2.101/jjm_packages
```

jjm2473, 2021-03-25:
@WangLeiCloudcade 你给的地址不能下载到原始文件啊，只是个github的页面，你试试这个就知道了 https://github.com/jjm2473/cdn/tree/master/openwrt/release/18.06.9/targets/realtek/rtd129x/packages/Packages.sig

所以还是只能用  raw.githubusercontent.com

jjm2473, 2021-03-25:
@WangLeiCloudcade 我前面提供的地址 https://github.com/jjm2473/cdn/tree/master/openwrt/release/18.06.9/packages/aarch64_cortex-a53/base ，是给人看的，不是给机器看的。软件源需要配置成机器看的地址。

jjm2473, 2021-06-27:
#9

closed 2021-06-27 by jjm2473

susie1986, 2022-02-05:
搞不定，科学上网也不行