jjm2473-org/openwrta #14: 再你的openwrt里重置了一下，结果再也进不去了！

beizi2, 2021-02-25:
各位莫要在openwrt里重置数据，一定要在安卓里重置，不然会再也进不去这个第三方openwrt，而且咋刷都不行，只能刷官方固件后才可以进官方openwrt！

jjm2473, 2021-02-25:
我确实也没试过在op里面重置。

你说的后面怎么刷都进不去，是因为我的op跟官方op使用的数据区不一样，如果没启动我的op，是没办法在安卓里面重置我的op的。要想在无法启动我的op的情况下重置，参考这个链接 https://github.com/jjm2473/openwrt/wiki/RTD129X-Openwrt-18.06#%E9%87%8D%E7%BD%AEopenwrta ，刷入一个重置包就行了。

beizi2, 2021-02-27:
试了刷重置包，没用

beizi2, 2021-03-01:
> 我确实也没试过在op里面重置。
> 
> 你说的后面怎么刷都进不去，是因为我的op跟官方op使用的数据区不一样，如果没启动我的op，是没办法在安卓里面重置我的op的。要想在无法启动我的op的情况下重置，参考这个链接 https://github.com/jjm2473/openwrt/wiki/RTD129X-Openwrt-18.06#%E9%87%8D%E7%BD%AEopenwrta ，刷入一个重置包就行了。

试过所有的方法了，你的重置包还有zidoo的强刷完整包降级重置都不行！大哥要是有空给研究一下呗，我实在是没招了！

tommytianhm, 2021-03-18:
我前几天还正常，今天重刷死活都不能进后台了，重置也没用。并且进Android后USB口不能识别外挂硬盘。

jjm2473, 2021-03-19:
@tommytianhm 试了官方的强刷教程吗，如果官方强刷以后还是有问题，那就可能是flash坏了

tommytianhm, 2021-03-19:
我就是强刷救回盒子的，flash应该没问题，（有坏道？）。再试着刷这个固件再也没成功过，再次通过会滚官方openwrt救回。这怎么破？

jjm2473, 2021-03-21:
@beizi2 @tommytianhm 
固件更新了 https://github.com/jjm2473/openwrt/wiki/RTD129X-Openwrt-18.06#openwrta-1806-r210320

刷新的固件，还有新的重置包，就能修复了。新固件支持在页面上重置。

closed 2021-06-27 by jjm2473