jjm2473-org/openwrta #18: 我是用的一个外贸盒子RTD1295方案，无法OTA升级，请问如何刷入这个OpenWRTA

wg-cn, 2021-04-21:
你好，
我是用的外贸盒子，叫Xnano X5，用的是RTD1295+2G+16G方案，配置跟X9s几乎一样；
我在OTA页面无法选择zip文件进行OTA升级，请问这种情况如何刷入OpenWRTA？
谢谢

jjm2473, 2021-04-22:
能进刷机模式吗？就是开机时按住install键或者reset键。

wg-cn, 2021-04-22:
> 能进刷机模式吗？就是开机时按住install键或者reset键。

可以进去，我这个自己重置刷过几次img恢复文件，同时按住开机和reset键

jjm2473, 2021-04-22:
现在的openwrt的ssh能连上吗，看看分区。用命令 `blkid` 应该就行

wg-cn, 2021-04-22:
> 现在的openwrt的ssh能连上吗，看看分区。用命令 `blkid` 应该就行

可以连上SSH，我晚上回家试下，连ssh的各项功能之前试过正常，blkid命令查看完，下一步如何操作，谢谢

jjm2473, 2021-04-22:
下一步我会提供刷机的img，以及用法

wg-cn, 2021-04-22:
> 下一步我会提供刷机的img，以及用法

感谢大神，我之前自己提取了zidoo的img能刷进去，就是很多系统软件出现FC，自己也没有技术搞编译，只能求助大神了，谢谢

jjm2473, 2021-04-22:
如果你的设备分区跟zidoo的一样，那么

参考这个 https://jjm2473.github.io/nas/2020/08/26/rtd129x-openwrt-1806.html#%E5%9B%9E%E6%BB%9A 

虽然是回滚到官方openwrt的教程，实际上把教程中“官方OTA包”换成“我的OTA包”，就是变成刷我的openwrta了。

wg-cn, 2021-04-22:
> 如果你的设备分区跟zidoo的一样，那么
> 
> 参考这个 https://jjm2473.github.io/nas/2020/08/26/rtd129x-openwrt-1806.html#%E5%9B%9E%E6%BB%9A
> 
> 虽然是回滚到官方openwrt的教程，实际上把教程中“官方OTA包”换成“我的OTA包”，就是变成刷我的openwrta了。

您的OTA包是只有openwrta的zip文件，官方的刷机包是img文件，按照回滚的流程，我是要把官方img文件里的openwrt文件替换成你的ota包的zip文件吧？

jjm2473, 2021-04-22:
嗯

wg-cn, 2021-04-22:
> 现在的openwrt的ssh能连上吗，看看分区。用命令 `blkid` 应该就行
使用命令后显示如下，不知道跟x9s的分区是啥样的

root@OpenWrt:~# blkid
/dev/mmcblk0p1: UUID="57f8f4bc-abf4-655f-bf67-946fc0f9f25b" TYPE="ext4"
/dev/mmcblk0p2: UUID="57f8f4bc-abf4-655f-bf67-946fc0f9f25b" TYPE="ext4"
/dev/mmcblk0p3: LABEL="cache" UUID="f73de97f-7017-ca5e-9b47-ca9c8f633e83" TYPE="                                                             ext4"
/dev/mmcblk0p7: UUID="57f8f4bc-abf4-655f-bf67-946fc0f9f25b" TYPE="ext4"
/dev/mmcblk0p9: TYPE="squashfs"
/dev/mmcblk0p10: LABEL="nasetc" UUID="f4c72ecc-3e10-d251-8d3e-65c472a903a5" TYPE                                                             ="ext4"
/dev/zram0: TYPE="swap"
/dev/mmcblk0: PTTYPE="dos"

jjm2473, 2021-04-23:
分区没问题。不能访问的话，我也没办法判断哪里出问题。如果你能用ttl，跟踪下开机的日志，才能进一步分析

wg-cn, 2021-04-24:
> 分区没问题。不能访问的话，我也没办法判断哪里出问题。如果你能用ttl，跟踪下开机的日志，才能进一步分析

好的，谢谢大佬，我看下怎么搞  买了ttl接头和杜邦线

jjm2473, 2021-04-24:
哦，安卓能启动的话，adb连上去它的shell，执行`dmesg`命令或者`cat /proc/kmsg`可以查看内核的日志，刚开机的话也许能看到哪里出问题

wg-cn, 2021-05-10:
> 哦，安卓能启动的话，adb连上去它的shell，执行`dmesg`命令或者`cat /proc/kmsg`可以查看内核的日志，刚开机的话也许能看到哪里出问题

我用大佬的上一个移植的项目按照回滚的方式把“RTD1295 OpenWrt 19.07.0-rc2+ 移植项目”刷入我的盒子，是可以启动Openwrt的，就是更新软件会报一堆错误

jjm2473, 2021-05-12:
更新软件报错是网络问题，需要全局翻墙才能下载软件包

wg-cn, 2021-05-12:
> 更新软件报错是网络问题，需要全局翻墙才能下载软件包

好的，谢谢。不过我重启盒子之后又进不去openwrt了，安卓系统进入openwrt设置会FC，估计还是不兼容

jjm2473, 2021-05-12:
只能看ttl日志了，不然不知道发生了什么

whp1942, 2021-05-19:
> 你好，
> 我是用的外贸盒子，叫Xnano X5，用的是RTD1295+2G+16G方案，配置跟X9s几乎一样；
> 我在OTA页面无法选择zip文件进行OTA升级，请问这种情况如何刷入OpenWRTA？
> 谢谢

大神，我有个矿机主板CPU也是RTD1295的，现在用ssh刷了15.05版，能否把你的源地址给我，我单独刷刷你的OPenwrt的，我看这个主板也应该是安卓+OP的，但是进不去安卓，我自己单独刷的OP，能否把你的源地址给我一份，我单独刷你的op，下面这个是我刷的OP
修改包地址

cat <<EOF > /etc/opkg/distfeeds.conf
src/gz chaos_calmer_base http://archive.openwrt.org/chaos_calmer/15.05/arm64/generic/packages/base
src/gz chaos_calmer_luci http://archive.openwrt.org/chaos_calmer/15.05/arm64/generic/packages/luci
src/gz chaos_calmer_management http://archive.openwrt.org/chaos_calmer/15.05/arm64/generic/packages/management
src/gz chaos_calmer_packages http://archive.openwrt.org/chaos_calmer/15.05/arm64/generic/packages/packages
src/gz chaos_calmer_routing http://archive.openwrt.org/chaos_calmer/15.05/arm64/generic/packages/routing
src/gz chaos_calmer_telephony http://archive.openwrt.org/chaos_calmer/15.05/arm64/generic/packages/telephony
EOF
修改包架构文件
vi /etc/opkg.conf
按i键再删掉（意思可以修改再继续操作）
输入这三行
arch all 100
arch rtd1295 200
arch arm64 300
按ESC Shift+: 输入wq回车
更新包
opkg update
删除原界面
opkg remove luci*
此处运行3次
重新安装新界面
opkg install luci

jjm2473, 2021-05-19:
你上面执行之后有啥问题？

我的15的源在
https://github.com/jjm2473/rtd1295-blob

whp1942, 2021-05-19:
QB我自己都装不上，下载的话，一会就断网了，所以想换换看看

https://github.com/jjm2473/rtd1295-blob

jjm2473, 2021-05-19:
qb估计不能直接安装，应该需要安装musl libc，但是rtd的op里面本来就有glibc了，装多个libc可能会有冲突导致大部分程序无法运行，所以还是别搞了

whp1942, 2021-05-20:
我是看网上说的，别人有装QB的，下载也能达到8--10兆，但是一会就掉线，必须重启设备网络才能好。这个可能就是你说的原因，装QB后会有冲突，部分程序无法运行，导致断网。这个设备是  星季云 ， 可能这个设备的量太小了，没有大神去研究刷固件，买这个矿渣的原因是因为好看带路由功能还能带4个硬盘（虽然是USB转的）。原矿机星季云用的就是RTD1295的CPU，应该是跟Zidoo X9S一样的配置，都是原机自带安卓+OP的，矿渣带原公司的固件，只能看见普通路由器的界面

jjm2473, 2021-05-20:
掉线跟libc没啥关系，libc错误的话，有些程序直接不能启动了，不会在运行中才挂掉。掉线可能是内存不足

whp1942, 2021-05-20:
现在刷完OP后，我自己的看内存是有两个G是满的，可能这2个多G装的安卓系统，照理来说剩下1-2个G安装OP在加个下载应该够用了，不会出现掉网络的

jjm2473, 2021-05-20:
安卓系统比较耗内存。看看qb占多少内存吧，实在不行试试transmission，反正都能下载bt

closed 2021-09-26 by jjm2473