jjm2473-org/openwrta #19: Aria2下载没有速度

pipidaxian, 2021-06-14:
我的Aria2用不了，设置我都看过一切正常，但是用链接下载会显示error:name resolution failed，用BT下载会在硬盘创建初始文件，但是一直没速度（加了tracker也一样）。可能是什么问题呢？

jjm2473, 2021-06-15:
name resolution failed，应该是DNS问题吧，你用的wifi吗，是不是静态ip

pipidaxian, 2021-06-26:
我是用的z9s接的我的华硕路由器WiFi，在openwrt这里没有额外设置过网络。openwrt这里的DHCP和静态IP我看全是空的没有配置，要怎么弄下吗？

jjm2473, 2021-06-26:
openwrt的网络依赖android那边的网络配置，不是自己控制的。我问的是android那边怎么连接网络的，用的wifi吗，是不是静态ip

jjm2473, 2021-06-26:
你在openwrt里面ping qq.com试试

pipidaxian, 2021-06-26:
用的是WiFi，自动从主路由器获取的IP地址。我在openwrt里ping了下试试，ping任何网站都ping不出来，好像openwrt没有连上外部网络。需要如何设置呢？

jjm2473, 2021-06-27:
openwrt不是自己控制网络的，只要android连上网，op就能上网。但是DNS配置不是网络层的东西，是libc层处理的，op虽然连上网了，但是DNS可能没配置，那就没办法进行域名解析了。你试试ping 114.114.114.114

jjm2473, 2021-06-27:
如果能ping通114.114.114.114，那肯定是DNS问题。执行 cat /etc/resolv.conf 看看

pipidaxian, 2021-06-27:
能ping通114.114.114.114，我在SSH执行了cat /etc/resolv.conf，什么也没发生，我用vim确认了下，发现resolv.conf文件是空的，啥也没有

jjm2473, 2021-06-27:
那就是android那边没来更新DNS。你的安卓固件是什么版本

jjm2473, 2021-06-27:
安卓的wifi配置页面截图看看

pipidaxian, 2021-06-27:
安卓系统的版本是Z9S的V2.3.35，截图如下：
[image]

[image]

[image]

jjm2473, 2021-06-27:
试试有线连接，关掉wifi，看DNS会不会正常

jjm2473, 2021-06-27:
ok，我自己测试出来了，wifi连接时DNS会空

pipidaxian, 2021-06-27:
感谢！原来是这样，那有别的办法把DNS加入到WiFi连接设置里吗？我的设备因为考虑要连接投影仪，没法挪动位置用有线。

jjm2473, 2021-06-27:
这个BUG是安卓导致的。现在的DNS配置不是op自己写的，而是安卓通过远程调用写进去的，有线的时候正常写入了，不知道为何无线就没写。我看看有没有什么错误吧

pipidaxian, 2021-06-27:
好的好的，谢谢了。我好像依稀记得刷您的op固件之前的z9s自带op固件好像可以ping外网，但是比较久了记不清了，算是提供个线索给您吧。我自己也再测试下

jjm2473, 2021-06-27:
嗯，因为以前是op那边也发起了DHCP请求，后来我看官方的安卓已经会自动更新op的DNS了，所以我就去掉了DHCP，改用官方的方案，但是没想到官方只给有线网络更新DNS，无线就直接空了。真TM坑爹

jjm2473, 2021-06-27:
我看了安卓的日志，确实没调用更新DNS的脚本。有线连接时会调用

pipidaxian, 2021-06-27:
哈哈哈坑爹，那我先暂时退回官方的，等您这边修复了再来刷您的 : )

jjm2473, 2021-06-27:
官方的用wifi可以正常DNS吗

jjm2473, 2021-06-27:
如果能用的话，到时 cat /etc/resolv.conf 看看

pipidaxian, 2021-06-27:
我已经退回了，看了下，官方的resolv.conf有三个nameserver，127.0.1.1，223.5.5.5，8.8.8.8。感觉你的固件是不是这三个加进去就好了。

jjm2473, 2021-06-27:
这三个是默认的，连接有线会被修改。而且这三个也不是路由器分发的DNS地址，只是个公共DNS而已

pipidaxian, 2021-06-27:
这样啊，那resolv.conf里除了这三个没别的了，可能在别的地方设置的吧

jjm2473, 2021-06-27:
我是说，如果你用wifi连接的话，那得到的DNS不是你的路由器配置的

jjm2473, 2021-06-27:
安装这个试试 https://github.com/jjm2473/RealtekOpenwrtDNS/releases/tag/v1.0.0

pipidaxian, 2021-06-27:
好的

jjm2473, 2021-06-27:
看这个吧 https://github.com/jjm2473/openwrt/issues/20

closed 2021-06-27 by jjm2473