jjm2473-org/openwrta #25: OpenwRTA refuses to load after months of use

HaoSs07, 2021-11-28:
Hello.
 I came to you with a very strange problem, OpenwRTA just started crashing on me ( random reboots) after around 4 months of working with it, I shut down the box, next morning i tried again and it was just dead. The interface does not load, unable to connect to it with SSH, the box boots just fine in android but the HDD storage is not detected.
I have tried from android openwrt reset to flash reset, total reflash, Zidoo OTA upgrade + OpenwRTA reflash, zidoo image recovery + OpenwRTA reflash, no luck. The original rtkwrt load just fine, but i soon as i flash OpenwRTA over it, can't access it anymore. Any idea on what i can try to make it work again ? Or the location to logs so we can see what the problem with OpenwRTA is ?

Thanks for your time.

jjm2473, 2021-11-29:
Did you try flashing https://jjm2473.github.io/assets/files/reset_openwrta.img ?  

see https://jjm2473.github.io/nas/2020/08/26/rtd129x-openwrt-1806.en.html#reset-openwrta

HaoSs07, 2021-11-29:
yes, that's the second thing i tried, the " flash reset " part.

jjm2473, 2021-11-30:
What the version of OpenwRTA you are using?  Upgrade to [OpenwRTA 18.06 R21.03.20](https://jjm2473.github.io/nas/2020/08/26/rtd129x-openwrt-1806.en.html#download) ,  this version fix the resetting function.

HaoSs07, 2021-11-30:
i was on the latest version. R21.03.20, i did a reset from the luci interface once it started crashing, but after I did all my settings it started crashing again ( crashes may be related to other things tho. qBittorrent was hanging and my HDD was very hot ) anyway, i also tried an older version R20.12.28, but had the same response as R21, no web interface, no ssh connection, and android throwing " unknown error " when trying to access storage

jjm2473, 2021-11-30:
OK, I can't figure out why resetting failed. R21.03.20 with reset_openwrta.img should works.
Did you upgrade android before OpenwRTA goes wrong?

What device you are using?
Can you use TTL/UART to get boot log?

HaoSs07, 2021-11-30:
No android upgrade, the device is a zidoo X10 ( same as x9s but has the internal HDD slot )
Never used TTL/UART but will see if i have all i need to try it.

jjm2473, 2021-11-30:
I don't have x9s or x10, for TTL/UART stuff you should get help from http://forum.zidoo.tv/index.php

HaoSs07, 2021-12-03:
I ended up getting a USB3.0 to 3.5" and connecting the HDD directly to a pi4, setting up qbittorrent in docker and it runs much much better, i always had problems with peer connections when using torrenting on the zidoo box, i will probably retire the x10, shame, it still has good hardware in it but the software lacks big time.
Anyway, thanks for your work on this openwrt port, i still got to use it for half a year, much better then what zidoo ever did.

closed 2021-12-03 by HaoSs07