jjm2473-org/openwrta #26: ota升级无效

lixiongjiu, 2021-12-02:
大神，用了你最新的固件进行了升级，每次过程都正常，但是重启后发现还是15版本，升级无效

jjm2473, 2021-12-03:
啥设备啊

lixiongjiu, 2021-12-03:
> 

Z9s，升级前版本是自带openwrt ，版本15.X
升级方式：MAC下载固件，192.168*登录浏览器，界面升级，过程无报错，但是升级后发现还是原版本
请问你是怎么升级的？

lixiongjiu, 2021-12-03:
> 啥设备啊

Z9s，升级前版本是自带openwrt ，版本15.X
升级方式：MAC下载固件，192.168*登录浏览器，界面升级，过程无报错，但是升级后发现还是原版本
请问你是怎么升级的？

jjm2473, 2021-12-03:
你都没看安装说明，下载的OTA包是在安卓系统里面刷的，就跟官方的安卓固件一样。这个OTA包是安卓固件的格式，openwrt里面是认不了的

lixiongjiu, 2021-12-03:
谢谢指导发自我的华为手机-------- 原始邮件 --------发件人： 练亮斌 ***@***.***>日期： 2021年12月3日周五 16:47收件人： jjm2473/openwrt ***@***.***>抄送： lixiongjiu ***@***.***>, State change ***@***.***>主    题： Re: [jjm2473/openwrt] ota升级无效 (Issue #26)
你都没看安装说明，下载的OTA包是在安卓系统里面刷的，就跟官方的安卓固件一样。这个OTA包是安卓固件的格式，openwrt里面是认不了的

closed 2021-12-04 by lixiongjiu

luckiestone, 2023-06-19:
遇到了同样的问题，设备X10，安卓页面本地升级，升级成功后重启，检查web页面仍然为15版本。