jjm2473-org/openwrta #32: 关于Z9S升级固件失败的问题

zhiyong202, 2022-07-13:
首先感谢大神提供的固件，之前一直很稳定。
后来可能是Z9S自动升级了，openWrt无法进入。
因此通过官方教程刷新到了[v2.3.88](https://www.zidoo.tv/Support/downloadList/target/n9KLv4h9WZZKKmVViAFMcQ%3D%3D.html#)
但是在刷OpenwRTA 18.06 R21.03.20时，一直无法进入到安卓系统。
想请教下是不是不支持最新的官方固件。能支持的官方固件有哪些？

jjm2473, 2022-07-14:
R21.03.20是21年3月更新的，官方v2.3.88是今年才更新的，如果官方确实改了什么导致无法启动，那只能我这边适配，或者使用旧版的官方固件。如果我来适配，不知道老的官方固件能不能正常启动

zhiyong202, 2022-07-14:
我也试了官方20年的固件2.3.35，同样无法进入安卓系统。 想问下R21.03.20支持从哪个官方固件升级。

jjm2473, 2022-07-14:
那估计不是因为官方固件升级导致启动不了。
参考这个教程“重置openwrta”
https://github.com/jjm2473/openwrt/wiki/RTD129X-Openwrt-18.06#%E9%87%8D%E7%BD%AEopenwrta

jjm2473, 2022-07-14:
> 想问下R21.03.20支持从哪个官方固件升级。

R21.03.20支持21年3月20之前的固件

jjm2473, 2022-07-14:
新的官方固件还没测试过

zhiyong202, 2022-07-17:
> 

确实是openwrta导致启动不了。 
因为之前老版本重置了openwrta， 我以为重刷了官方固件所有设置都清空了。 
重置之后， 官方v2.3.88 + openwrta R21.03.20工作良好，感谢~

jjm2473, 2022-07-17:
openwrta的数据不在官方固件的范围内，所以不会被官方的重置方法重置掉

closed 2022-07-17 by jjm2473