jjm2473-org/openwrta #33: 软件包无法opkg update

pipidaxian, 2022-08-13:
在ssh界面输入opkg update后无法更新，显示
Collected errors:
 * opkg_download: Failed to download https://mirrors.bfsu.edu.cn/openwrt/releases/18.06.9/targets/realtek/rtd129x/packages/Packages.gz, wget returned 5.
 * opkg_download: Failed to download https://mirrors.bfsu.edu.cn/openwrt/releases/18.06.9/packages/aarch64_cortex-a53/base/Packages.gz, wget returned 5.
 * opkg_download: Failed to download https://mirrors.bfsu.edu.cn/openwrt/releases/18.06.9/packages/aarch64_cortex-a53/luci/Packages.gz, wget returned 5.
 * opkg_download: Failed to download https://mirrors.bfsu.edu.cn/openwrt/releases/18.06.9/packages/aarch64_cortex-a53/packages/Packages.gz, wget returned 5.
 * opkg_download: Failed to download https://mirrors.bfsu.edu.cn/openwrt/releases/18.06.9/packages/aarch64_cortex-a53/routing/Packages.gz, wget returned 5.
 * opkg_download: Failed to download https://mirrors.bfsu.edu.cn/openwrt/releases/18.06.9/packages/aarch64_cortex-a53/telephony/Packages.gz, wget returned 5. 
请问这是无法连接软件包吗？但是我上层的路由器已经开了科学上网，应该所有设备都翻出去了才对啊

pipidaxian, 2022-08-13:
哦，我看了下，好像只有这个国内镜像源有这个问题，github的反而正常的，不知道咋回事。另外问下，可以安装python3吗？

jjm2473, 2022-08-16:
应该只是网络不稳定或者系统时间不对，还有建议使用腾讯的镜像仓库

jjm2473, 2022-08-16:
https://mirrors.bfsu.edu.cn/openwrt/releases/18.06.9/targets/realtek/rtd129x/packages/Packages.gz 肯定是不存在的，中国错误忽略

closed 2022-08-16 by jjm2473