jjm2473-org/openwrta #39: 升级OpenwRTA ，网络出现故障。

xianxin21, 2023-03-26:
升级OpenwRTA 18.06 R23.01.12，安卓端可以访问网站，网络也是通的。OpenwRTA端不能访问网站，ping网站不通，ping ip地址可以。重置后仍然不行。尝试过的方法：1.安卓端设置静态ip，Dns设置本地的，或阿里，114都设置过，问题依然存在。
2.安卓端设置DHCP获取，问题依然存在。3.OpenwRTA端，LAN设置静态ip，也不行

[image]
[image]
[image]

jjm2473, 2023-03-28:
https://github.com/jjm2473/openwrta/wiki/RTD129X-Openwrt-18.06#%E9%A2%98%E5%A4%96%E8%AF%9D

closed 2023-03-28 by jjm2473

xianxin21, 2023-04-12:
我是用的网线直联出的问题，打APP补丁也没有用。
尝试过用无线wifi模式打APP补丁问题就不存在。
还望大神指教！

jjm2473, 2023-04-12:
openwrt端不能配置任何网络，以免跟安卓冲突。

jjm2473, 2023-04-12:
要么你就设置个固定的dns，反正现在只是dns有问题。

xianxin21, 2023-04-12:
openwrt端重置过，也不行。安卓端、openwrt端都试过配置过dns，也不行。黔驴技穷了。

jjm2473, 2023-04-12:
你可以把openwrt的/etc/resolv.conf删掉，然后重新创建一个，按linux的方式填写就行了。

xianxin21, 2023-04-12:
好的。非常谢谢。我试试。

xianxin21, 2023-04-12:
此方法有效，重启后又失效，需要重新删除文件后，重建文件添加内容。
总结一下：
1.无线wifi下，打了APP补丁，DNS不会出问题，重启也没有问题；
2.有线网络自定义指定IP地址，指定DNS会出问题，更新不到OP端，查看OP端的/etc/resolv.conf  中的nameserver是我的网关地址，理论上网关地址也可以解析的，但问题依然存在。但按照楼上大神方法可以治标，但治不了本，重启后又不行，nameserver又回到了我的网关地址，需要按上述流程再来一遍。
3.总的来说，算基本解决当前困惑，感谢大神耐心指点。

jjm2473, 2023-04-12:
那简单，你加个启动脚本不就行了，openwrt的启动项那边就有个写启动脚本的

xianxin21, 2023-04-13:
还要麻烦大神指教一下代码。

jjm2473, 2023-04-13:
```
rm -f /etc/resolv.conf
echo >/etc/resolv.conf
echo "nameserver 223.5.5.5" >>/etc/resolv.conf
```

xianxin21, 2023-04-13:
大神给力，回复迅速。再请教一下，可能不涉及您的领域，安卓界面下，原装红外遥控器红色关机键没有用，无法关机，之前原厂版本可以。后面安卓界面也升级到最新了。

jjm2473, 2023-04-14:
你这个啥设备啊，我都没遇到过这些问题

xianxin21, 2023-04-14:
> 你这个啥设备啊，我都没遇到过这些问题

我的是x9s，遥控器按键有红灯亮，但没有反应，无法关机。x9s外接上了移动硬盘保持PT在用。但拔掉硬盘也还是不行。
只能断电关机，或通过op端重启。

jjm2473, 2023-04-14:
是不是你自己设置了遥控器按钮功能，z9s是可以自定义关机键的功能的