jjm2473-org/openwrta #41: Unable to add software source issue

EugeneZengXY, 2023-06-19:
I am a domestic user and cannot add custom software sources because I need to install tailscale and SSR. Can the author assist with remote assistance? Sunflowers or todesks are both acceptable，my wechat is 1149379753

jjm2473, 2023-07-01:
You need to add aarch64-cortex-a53 source. If some app run in trouble, maybe you are using uncompatible source.

official repo: https://github.com/jjm2473/cdn/tree/master/openwrt/release/18.06.9

if you want to compile packages by yourself, check https://github.com/jjm2473/openwrta/blob/openwrt-18.06/.github/workflows/ci.yml

jjm2473, 2023-07-01:
如果是国内用户，就用中文吧，我的英文可不好。软件包需要是 aarch64_cortex-a53 架构的，而且不要觉得随便找个软件源就所有软件包都能用，毕竟这个固件还是op18，很老的版本了。