jjm2473-org/openwrta #43: 更新到18.06.9的新版openwrta安装istore缺少luci-lib-ipkg依赖

davidsung1983, 2023-07-14:
更新到18.06.9的新版openwrta安装istore缺少luci-lib-ipkg依赖
大佬还更新固件吗？能不能把istore和homeassistant也集成进去？
能力有限实在不会搞：（

jjm2473, 2023-07-14:
你可以试试这个脚本能不能安装istore：
```
opkg update || exit 1
cd /tmp
wget https://github.com/linkease/openwrt-app-actions/raw/main/applications/luci-app-systools/root/usr/share/systools/istore-reinstall.run
chmod 755 istore-reinstall.run
./istore-reinstall.run
```

homeassistant需要docker，这个固件是不可能支持docker的，因为内核使用的是官方固件自带的，内核太老且缺少一些功能。

davidsung1983, 2023-07-14:
应该不需要docker吧，docker版本缺太多东西了。https://github.com/gl-inet/home-assistant-on-openwrt 按照这个试了一下，需要python3.7,这个只能到3.6。。。
istore也缺依赖，说是19以上的版本才有。。。好像只能自己编译到固件里了，用luci-base替代luci-lib-ipkg
主要还是不太会，linkease的istoreOS官方的板子就是RTD1296的。

jjm2473, 2023-07-14:
istoreos就是我开发的啊，但是ars2使用的是自己的内核，不是安卓的内核，自己的内核当然可以支持docker

davidsung1983, 2023-07-15:
看到了，istore也是你发布的。果然厉害。
你是说istore也需要dorcker支持？
不是说可以编译到openwrt固件里吗？脚本是安装不下去的。我把IPK传到openwrt里安装的，最后安装istore的时候提示缺少luci-lib-ipkg依赖。。。

jjm2473, 2023-07-17:
istore本身不需要docker支持。如果你确定不需要luci-lib-ipkg（可能老的luci还没把这个功能独立出来），那就强制安装啊，opkg命令不是支持--force-depends吗