jjm2473-org/openwrta #44: 用了差不多一周的几个问题

davidsung1983, 2023-07-19:
1、https访问报安全提示。。。有没有哪里可以关掉的？
2、SMB共享会掉，我是X20的机器，挂了2个3.5的盘，有时候SMB共享会掉，重启一下又好了，有时候又会掉一个。不知道啥原因；
3、X20后置有两个个USB3.0的A口和C口，无法识别固态盘，但是可以识别2.5寸的机械盘，不知道和固件有没有关系；

jjm2473, 2023-07-20:
首先，这固件只在x9s和z9s上测试过，你要用在其他设备上，出问题也很正常。其次，这固件也不是完整的固件，不包括内核，只是一个rootfs，内核相关的故障跟这个固件无关，例如usb什么的。

https默认不是强制的，你用http就没安全提示了。如果你要用https，那就要自己制作服务端证书，否则浏览器会报安全错误，注意这是浏览器的安全机制，跟固件无关！

BladPit, 2023-08-08:
没想到大佬还在更新这个项目，之前用了挺长时间，现在转NAS了，感谢你的无私贡献

> 首先，这固件只在x9s和z9s上测试过，你要用在其他设备上，出问题也很正常。其次，这固件也不是完整的固件，不包括内核，只是一个rootfs，内核相关的故障跟这个固件无关，例如usb什么的。
> 
> https默认不是强制的，你用http就没安全提示了。如果你要用https，那就要自己制作服务端证书，否则浏览器会报安全错误，注意这是浏览器的安全机制，跟固件无关！

jjm2473, 2023-08-11:
已经挺长时间没更新了，常用的软件没问题就行。项目有自动编译的功能，只是小改的话还好

jyothis89, 2025-05-27:
> This project is no longer maintained. As long as the kernel version of the
> Android firmware does not change, the OpenWRT firmware released here will
> still work. The kernel is unlikely to change, because Realtek has long
> since stopped supporting RTD129X.
>
> The latest firmware can be downloaded from here
> https://jjmgit.my.to/nas/2020/08/26/rtd129x-openwrt-1806.en.html#download

Thank you 👍